AllYarnsAreBeautiful/ayab-firmware #2: KG882 Firmware fixen

chris007de, 2016-11-26:
_From @niccokunzmann on June 7, 2016 19:30_

siehe #42, KG882 Option einfügen

_Copied from original issue: AllYarnsAreBeautiful/ayab-desktop#43_

chris007de, 2016-11-26:
falsefully closed due to rebasing (duplicate issue # in commit messages)

chris007de, 2016-11-26:
_From @niccokunzmann on August 9, 2016 22:13_

@chris007de is this still open and should we move it to the firmware repository?

chris007de, 2018-02-20:
Starting to knit from either side: https://github.com/AllYarnsAreBeautiful/ayab-desktop/issues/247

chris007de, 2018-03-25:
Implemented in a36dc833ce8620087e0ada1b0dcd2651a17c9a94 and ready to be tested.

chris007de, 2018-04-17:
Written by @Adrienne200 on 2018-04-06:

> I’m not getting the garter carriage to work very well. Did you get any feedback from the other testers yet?
KG88 (no extra turnaround cams), 910 machine, EMSL board.
Default settings except aligned on the left, my 60 stitch Triangles picture, L30-R30.
Start the garter carriage on the left.
First row doesn’t do the first few knit (g-carriage) stitches. Then it does do the second half of the row correctly.
2nd row (R - L) is perfect.
3rd and other L-R rows does the same, it fails to pattern the first half, (all purls) up to about L5, then it’s OK after that.
So the right hand half (as you look at the machine) looks good except that on row 5 there’s an extra set of 4 knits that aren’t supposed to be there. It happened again in the same place in the next repeat. I tried another sample  with center alignment which did the same.

> I notice that on the right, the beep happens as the whole garter carriage has moved off the working needles and is turning around.  But on the left, the beep happens much earlier, just as the needle has cleared the knitting, but half of the carriage still has a ways to move before it can turn around. But the move up to the next row happens as expected, I just thought I'd note this.

chris007de, 2018-04-17:
Written by @chris007de on 2018-04-09:

>> I’m not getting the garter carriage to work very well. Did you get any feedback from the other testers yet?
Actually, one user on FB did manage to do KG tests so far and is reporting a similar behavior.
Thanks for sending in log and pattern, I'll try to have a look at it asap.
>
> I notice that on the right, the beep happens as the whole garter carriage has moved off the working needles and is turning around.  But on the left, the beep happens much earlier, just as the needle has cleared the knitting, but half of the carriage still has a ways to move before it can turn around. But the move up to the next row happens as expected, I just thought I'd note this.
>
>> Which brings me to the other changes.
> I think Windell changed the beep time to 12 steps clear of the given work needles on both sides. (Previously 12 right and some higher number 32? on the left) In some brief testing with the K carriage I think that’s working. Though 12 may be too low, the beep can get lost in the sound of the carriage moving; we might consider experimenting with say 16?
>
>That's definitely caused by the change of the left hand offset from 32 to 12. On the left side, the machine is now thinking the line is finished 20 needles earlier than before.
This behaviour is currently the same for all carriage types. So if it is proofed that for K and L 12 is ok on both sides, but KG needs more, we could take this into account.
The change is described in https://github.com/AllYarnsAreBeautiful/ayab-firmware/issues/9, let's discuss this topic there.

chris007de, 2018-04-17:
Written by @Adrienne200 on 2018-04-10:

> Are you certain you built it right this time? Looking pretty much the same as before. :(  And yes, I double-checked that I had installed Post5, and reflashed the firmware. Windows 7.
>
>This time I noticed that it sometimes didn't beep and count up on the right. That part worked better if I had some extra needles in work on the right, (a 25 stitch pattern, centered, L25 - R30) Making the garter carriage walk those extra 5 stitches made it beep reliably on the right. Didn't improve the patterning any.
>
>If you're sure we're working from the same code and it should work, can you describe your setup where you got it working. What pattern, how many needles, what machine and OS?

chris007de, 2018-04-17:
Written by @chris007de on 2018-04-10:

>I'm pretty sure that it's the correct firmware packed in the setup now, but I will double check after work.
>
>Until then, you could try to use the test mode tab in the software to see if your KG is recognized as type G carriage when entering from the left. (when the left magnet passes the left sensor, first it should show L, then G).
If it shows K after the left magnet passed the left sensor, something is still broken (please check magnet orientiation, marking of both magnets should point inward afaik).
>
>I tested on Linux with our KH-910.
I tested with various patterns, e.g. https://github.com/AllYarnsAreBeautiful/ayab-patterns/blob/master/testPattern_40x40.png with default application settings.

chris007de, 2018-04-17:
Written by @Adrienne200 on 2018-04-11:

>My news still isn’t so good. I looked more closely at test mode, and my 910 is just not ever deciding that it’s a G carriage. I see the G flash for a moment as the second magnet moves past the left sensor, so I have the right firmware, but then it changes its mind and decides that it’s an L carriage. I’m open to the possibility of trying the adjustments for the Hall sensor as mentioned in the service manual, but not yet.
>
>I moved my EMSL board to a 930, and reflashed. On the 930, I got it to work once and let it run for a few rows. But then subsequent attempts all failed, back to the random-ish behavior when it thinks it has a K or L carriage.
>
>The 930 in Test mode was more informative, now I see how you intended it to work, as you described. Moving the G carriage across the left Hall sensor I can see it decide L, then K, the K again and finally G after both magnets have passed. (Where the 910 had flashed G then gone with L after the second magnet) So test mode on the 930 seemed to be reliably deciding it’s a G carriage. But trying to do an actual knit, only one successful attempt out of 6. And I only let it run for a couple of rows, so now I’m not 100% certain that it really did work that one time.
>
>So that brings up a question. What is your exact start-up procedure?
>
> - cast on in the middle of the needlebed and knit some rows with the K carriage.
> - put on the rails if needed
> - position the G carriage with both magnets outside the left turn mark
> - set up AYAB as usual, not Test mode, nothing special needed in the software. (Assuming 0.91 sw and fw)
> - Configure, Knit.
>Several possible ways to start:
> - Switch on the G carriage and let it walk to the middle by itself. Notice that the AYAB startup sound has already finished by the time the second magnet crosses the sensor. Does that matter? When does it decide for sure which carriage it is?
> - Lift the top half and slide the carriage onto the bed manually, quite quickly. Both sets of magnets have passed the sensor before the startup sound completes.
> - Same, but more slowly.
>Which do you do, and would you expect the others to work?

chris007de, 2018-04-17:
Written by @chris007de on 2018-04-11:

> My news still isn’t so good. I looked more closely at test mode, and my 910 is just not ever deciding that it’s a G carriage. I see the G flash for a moment as the second magnet moves past the left [...]
Good thing is that the Arduino is recognizing the K->L transition (which means G) on the left magnet.
I would say that it switches back to L shortly after deciding for G is a flaw in our carriage-decision, as it does not take hardware/carriage/magnet mounting tolerances into account.
>
>Currently, when the machine has decided for G, and another hall sensor reading comes in, it is currently not prohibited to decide for another type.
It seems that on our machine, the L hall reading which causes decision for G is the last hall reading of the left magnet, whereas on your machine, there still comes another L reading, which makes it change its mind again.
>
>So if my interpretation is right, this will definitely happen on other machines also, and we have to take care of it. I even already had this in mind when doing the first shot of G-carriage decision  -> https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/v0.95/encoders.cpp#L71
I think we have to implement a hysteresis for the carriage decision, e.g. when it has decided for G, it is not allowed to change its decision again for a movement of 1 or 2 needles, so it stays in G after the left magnet has left the left sensor when moving to the right.
>
>For a reliable implementation of the algorithm, it would be very helpful if I had the exact test mode readings of your carriage. So when moving to the right past the left sensor, how many needles does it show L (right magnet), how many K (right magnet) how many needles are in between the magnets, how many K (left magnet) and how many G/L (left magnet)?
It takes a few minutes to do this, but I think it is worth the effort.
>
>----
>
>Regarding KH930, we do not have any experience, as our KG88 is not mechanically compatible to the KH930 we own, so we were only able to read out the hall values, but could not knit so far.
>
>But to rule out any differences in the knitting process, here is how we knitted with KG on KH910:
>1. Cast on with K carriage
>2. Move K carriage out of the machine to the right
>3. Put on the rails
>4. Start AYAB, start a knitting process with an arbitrary pattern, all settings left on default
>5. Put KG on the left of the machine and let it move to the right by itself. Beep occurs some time after the first magnet was recognized. (Initialization does not care about the carriage type, only movement from left to right and presence of a magnet).
>6. Let the KG move to the right until the end-of-line beep occurs, switch direction, etc.
>
>So basically what you described, and the first choice of the possible ways to start.
>Manual movement of the KG (lifting the top) was not tried yet. Moving the carriage manually too fast might cause the Arduino to do an error in position counting, but "moderate" speed might be ok. As described above, it does not really matter when the beep occurs, initialization is happy when it sees a magnet moving from left to right.

chris007de, 2018-04-17:
Written by @chris007de on 20180-04-17:

>I think we should try to stay as close to the behavior of the original
machine as possible.
We did not know yet how they behave, so we implemented the easiest (but
probably not the most correct) way to recognize the carriage, which is
to use the left magnet.
We surely can also decide on the right magnet, it'll just take some more
code to ignore the left magnet - and do this right for both directions.
So we will have to take the current carriage position into account when
interpreting hall sensor measurements to be able to ignore the left
magnet when we know(/assume) a G carriage.
I'll try to think of some implementation for this and hope I will be
able to generate a new version before end of May.

dana80, 2019-04-06:
Hello I am a new user of Ayab on a KH910 and I tested the robot Kg88 with the map 1.01 of the Kh910 file. Does version 0.95 work well with the robot, because on the row from left to right it selects the needles only from 0 and knits normally from right to left?

duncan-d, 2019-04-15:
I just picked up a garter carriage yesterday. What options would you select in the GUI to run the garter carriage? Single-Bed 1 color? Or is the idea that it would be it's own Knitting Mode in the future?

Adrienne200, 2019-04-16:
I think that the existing basic 2-color single bed should be OK to start from, given that it already makes a half-hearted attempt to do the right thing. But if you decide that it has different timing and needs a mode of its own, that would work too. Or would it switch mode automatically under the hood?
During the runup to 0.95 there was a test mode which lets you see the signals as it passes the Hall sensors, so a good first step would be if you or Chris can turn that back on and do a test build, so you can get it to recognize the magnets and know that it's a garter carriage.

Adrienne200, 2019-04-16:
dana80 - as you can see the answer to your question is currently no, but now that duncan-d has a garter carriage to work with I hope there will be some movement.

dana80, 2019-04-16:
Merci pour votre réponse, oui je serai intéressée de faire fonctionner le KG 88 avec Ayab pour créer des nouveaux points. Je n'ai pas d'expérience sur la Kh910 seulement sur Erka25. Je viens de récupérer une 2eme Kh910 qui fonctionne encore electroniquement. Je vais

duncan-d, 2019-04-19:
How do you get into test mode? I feel like I've seen it talked about but never noticed how to do it myself. Also if someone could point me in the right direction to turn on the continuous status reporting that'd be much appreciated!

I did notice:

> That's definitely caused by the change of the left hand offset from 32 to 12. On the left side, the machine is now thinking the line is finished 20 needles earlier than before.
> This behaviour is [...]

I think the left offset was reduced to 12 so it would make more sense for everything else. Which could be one reason garter carriages aren't working well.

Adrienne200, 2019-04-19:
I expect you're correct that the G-carriage will need a different left offset. But the G-carriage has never worked, from long before that setting was changed. 
When I used a build with Test mode, I could see that mine wasn't reading the magnets correctly to switch to G-mode, it was in L, so it was trying to use the rules for the lace carriage. Maybe you could do an experimental build for yourself with the mode forced to G all the time?

Ilushock, 2020-02-10:
Ammmm..... And why it's closed? On this page https://ayab-knitting.com/ayab-software/ the same version of software (0.95). I even can not chek it with KG. In Win7 it doesn't work at all.

chris007de, 2020-02-10:
@Ilushock This issue is still open and work in progress...full KG support is expected to be available in AYAB v1.0, we're currently starting working on it.

Ilushock, 2020-02-11:
OK. Thanks.

roosdejuf, 2020-09-11:
Hello everyone,

Very new here, and I hope it's okay I write something here.. I have a Brother KH910 with AYAB shield, and have been using it for a little over a year now I think. I recently got a KG88II (although I haven't been able to get the magnets yet) but after reading up on the garter carriage support development so far, I came up with a few ideas. I am not a programmer, to say the least, my last experience was in high school, so my ideas might be completely useless. I doubted whether I should even comment at all, but I thought maybe some 'fresh eyes' might help something along. 

Ok, my ideas:

- From what I read, something seems to be 'off' with recognising which carriage it is. Would it be an option to leave the recognition not to the machine and software, but to the user? Maybe add a drop down menu option to go into Garter carriage mode? This could possibly interfere with the use of the normal K carriage and/or L carriage and the G carriage on the same project, unless there would be options to 'assign' rows or something like that? Or the option to switch modes while leaving the pattern and row memory intact? (Having the user manually pick a mode would maybe also be good for Lace, as the user could then just copy Brother patterns without having to remember to mirror them, as the Lace mode would do that automatically, but that's a whole different story). 

- From what I understood from the maintenance manual, in the original machine the garter carriage is recognised as such if two different magnetic poles are sensed within 13(?) needles of one another. How does the software determine the type of carriage? Maybe this maximum spacing would be useful (as it wouldn't be possible to have two magnet readings that close together otherwise even when using the K and L carriage simultaneously on the same bed)? Does the second set of magnets 'reset' something? How does the software know to ignore that second set, or shouldn't it ignore that?

- In Garter Carriage mode or if the machine keeps having problems with 'shifting' the rows, maybe the software could compensate for that? What happens if you let the machine decide whatever type it wants? Is it just a shifting over of the pattern?  On Ravelry someone managed to get a 150st wide sort-of-working situation using one magnet in the middle, but they also had to manually turn and restart the carriage every row. That person used an app to automatically offset the pattern. If a shift in the pattern is the only 'downside' of the software not properly assigning carriage type, that could maybe be automatically compensated for. 

These ideas are just the 'brain farts' I had while reading, and I realise that some of the things mentioned are less than ideal solutions. Through reading I've seen the dedication and hard work that has been put into this project for years, and I feel like you all want to deliver a product that is as good as possible. Maybe, though, a less-than-ideal solution would be very welcome as a temporary fix, as it may solve the problem for at least some of the projects. 

Finally, I would like to thank all of you for the hard work so far, AYAB is how I found machine knitting in the first place and I owe you all a lot for bringing me this option for creativity. Thanks so much.

Adrienne200, 2020-09-11:
Hi, and welcome to the project. I have been involved in this project off and on since 2015, and was the self-appointed chief tester for some previous releases. My coding skills are stuck at permanent beginner level so I'm not doing any of the programming, but I can test, so once there's a testable build of 1.0 I plan to get involved again.

As you see, a lot of recent awesome work has been done by t0mpr1c3, but he has not done any work yet with the garter carriage (or the lace carriage.) I'm not sure if he even owns a garter carriage. I've been hoping that as soon as he focuses in on this aspect he'll get the magnet recognition sorted out; it's not a super-complicated algorithm. And I agree that for testing, just putting it in the UI for the operator to choose would be a useful first step, then we'd decide later whether to leave it in the GUI or make it automatic only. I expect that there will also be timing and other special coding needed, but I think the basics of that is already in, though it may need some fine-tuning.

t0mpr1c3, 2020-09-11:
Thanks for those kind words, Adrienne. I do have a garter carriage, and will eventually get around to looking at all of this. I agree that it is an important feature for AYAB.

I have given a little thought to the idea that the user should explicitly specify which carriage they are using for each pass of over the needles. The next version of the software will be capable of that, although it might not actually be something that is actually implemented in the GUI. One reason to do this, as Sandy has mentioned, is that you might want to use more than one carriage.

Adrienne200, 2023-11-03:
The feature is at least partially implemented, even though it's not quite all there yet. Bugs about the detailed behaviors are being logged individually, so we can close this long-standing issue. Yay!

Adrienne200, 2023-11-03:
Close as implemented.

closed 2023-11-03 by Adrienne200