AllYarnsAreBeautiful/ayab-firmware #3: Improve serial protocol for firmware

chris007de, 2016-11-26:
_From @niccokunzmann on May 31, 2016 18:52_

find a way to specify how knitting instructions can be converted to instructions for the machine

_Copied from original issue: AllYarnsAreBeautiful/ayab-desktop#17_

chris007de, 2016-11-26:
_From @kirstin on May 31, 2016 18:55_

Documentation for this can be found here:  
[https://bitbucket.org/chris007de/ayab-apparat/wiki/english/Software](https://bitbucket.org/chris007de/ayab-apparat/wiki/english/Software)

chris007de, 2016-11-26:
Check also [this wiki entry](https://bitbucket.org/chris007de/ayab-apparat/wiki/english/Software/SerialCommunication) and its implementation in [ayab_communication.py](https://github.com/AllYarnsAreBeautiful/ayab-desktop/blob/master/software/python/ayab/plugins/ayab_plugin/ayab_communication.py)

chris007de, 2016-11-26:
_From @niccokunzmann on June 7, 2016 20:0_

- machine asks: give me line [uint8_t]
- serial answer: line number, 200bit as bit mask
- small and fast to be placed between interrupt
- checksum crc? unused
- needle position:
  - 1 = d
  - 0 = b
    laut andz

chris007de, 2016-11-26:
_From @niccokunzmann on July 24, 2016 12:49_

added the communication documentation http://ayabinterface.readthedocs.io/en/latest/communication/index.html

chris007de, 2016-11-26:
_From @niccokunzmann on July 24, 2016 13:27_

@chris007de @andz82 

waffle deleted my last post.
1. The debug message starts with "#" but in the specification with "0xFF" how to go about this? https://github.com/AllYarnsAreBeautiful/ayab-firmware/issues/1
2. cnfInfo [Chris' source code](https://github.com/AllYarnsAreBeautiful/AYABInterface/blob/communication-content/AYABInterface/communication.py#L77) does not match the [specification](http://ayabinterface.readthedocs.io/en/latest/communication/index.html#m4-c3)
3. [indState](https://github.com/AllYarnsAreBeautiful/AYABInterface/blob/communication-content/AYABInterface/communication.py#L89) is not listed
4. [in indState](https://github.com/AllYarnsAreBeautiful/AYABInterface/blob/communication-content/AYABInterface/communication.py#L91) what is the byte order of uint16?
5. [indInit](http://ayabinterface.readthedocs.io/en/latest/communication/index.html#m4-84) does not occur in the source code and in the sequence diagram - what does it do?

chris007de, 2016-11-26:
_From @niccokunzmann on July 24, 2016 15:30_

If you like you can add documentation here:
https://github.com/AllYarnsAreBeautiful/AYABInterface/blob/master/docs/communication/index.rst#the-reqstart-message

chris007de, 2016-11-26:
_From @niccokunzmann on July 24, 2016 16:50_

**Resolved (4)**  
In the [implementation](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/c236597c6fdc6d320f9f2db2ebeb17d64c438b64/knitter.cpp#L382), the higher order byte comes first.

**Resolved (5) and (3)**  
indInit does not appear in the source code. I assume that indInit means indState. Documentation needs to be adjusted. **Done**

chris007de, 2016-11-26:
_From @niccokunzmann on July 24, 2016 18:15_

**(1) and (2) Resolved**  
I assume I can update the specification according to what is used in the Arduino file.

I also linked [the documentation to the Python implementation and the Arduino source](http://ayabinterface.readthedocs.io/en/latest/communication/index.html#the-reqstart-message) code. #123 could improve linking.

chris007de, 2016-11-26:
_From @niccokunzmann on July 24, 2016 18:16_

@chris007de @andz82 You can review my assumptions.

chris007de, 2016-11-26:
_From @niccokunzmann on July 24, 2016 23:30_

New questions:
- (6) What is `initState` in `indState( bool initState = false)` ?
- (7) Do success messages use 0 == false and success != 0 ? I am not sure what [casting a bool to a byte](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/c236597c6fdc6d320f9f2db2ebeb17d64c438b64/knitter.cpp#L378) does. I would like to assume that success == 1 and all values >1 are invalid values. Before I decide on this, I would like to hear your thoughts. @chris007de 
- (8) What is [cnfTest](https://github.com/AllYarnsAreBeautiful/AYABInterface/blob/communication-content/AYABInterface/communication.py#L81) ?

**(6) Resolved**  
I assume that it means "success" != "0", failure == "0".

**(7) Resolved**  
**Update1** I think, based on [this answer](https://bytes.com/topic/c/answers/674697-bool-int-cast) that it is safe to assume 0 == false and success == 1 all other values are invalid.  
[Old text: I assume - because it is safe to assume - that 0 == false and success != 0 for all booleans in true/false messages in the specification. The specification needs to be rewritten then, but the Arduino program can be left untouched.]

**(8) Resolved**  
I assume:: cnfTest the message that should be answered to a [reqTest](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/c236597c6fdc6d320f9f2db2ebeb17d64c438b64/ayab.ino#L119). Both are not in the specification and should be put in the specification. @chris007de If you could write down what the test mode should do, this can be put in the specification.

chris007de, 2016-11-26:
Your assumptions are mostly correct.
Just some remarks to further clarify things:
- The host waits for a **indState(true)** message before requesting to start the knitting. On startup, the Arduino continuously checks for the initialization of the machine (carriage passed left hall sensor). When this happens, it sends an **indState(true)** to tell the host that the machine is ready to knit. After receiving this message, the host sends a **reqStart** message, which is immediately confirmed with a **cnfStart** message. When **reqStart** was successful, the Arduino begins to poll the host for line data with **reqLine**, the host answers with **cnfLine**. This reqLine/cnfLine happens each time the carriage moves passed the borders given by the Start/StopNeedle parameters in **reqStart**. When the host does not have any more lines to send, it marks the last line with the _lastLine_ flag in its last **cnfLine** message.
- reqTest (immediately confirmed by cnfTest) puts the Arduino in a Testmode, instead of a knitting mode (see https://github.com/AllYarnsAreBeautiful/ayab-desktop/blob/master/software/python/ayab/plugins/ayab_plugin/ayab_control.py#L605).
  Instead, it sends indState messages periodically, containing the sensor and position values.

I will clarify this in the specification today evening

chris007de, 2016-11-26:
_From @niccokunzmann on July 25, 2016 9:7_

@chris007de thanks. 
- (9) **Resolved** The sequence diagram does not contain the **indState** message. 
- (10) What does indState(false) mean?

**(9) Resolved**  
I assume the indState(true) message should be added to the diagram.

**(10) Resolved**  
**indState(initState)** is called with true/false. Thus, I assume that
- true means the controller has waited for this message and this is the first indState sent
- false means the controller can send this message from time to time to notify the host about changes.
  - This only takes place in the [Opstate_t s_test](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/c236597c6fdc6d320f9f2db2ebeb17d64c438b64/settings.h#L128). [Then `initState(false)` is called](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/c236597c6fdc6d320f9f2db2ebeb17d64c438b64/knitter.cpp#L298).

chris007de, 2016-11-26:
_From @niccokunzmann on July 31, 2016 21:41_

- (11) Is it possible to send a `reqStart` again to reset the start and stop needle in case the knitwork is not a rectangle? Will his require to reset the line count?
- (12) is it possible to know when the knitting process is over?

@chris007de @andz82

t0mpr1c3, 2020-08-26:
(11) This is not currently implemented.
(12) Yes, `cnfLine` includes a flag byte. If bit 0 is set, this signals the last line of the pattern.

I suggest to close this issue.

closed 2020-09-07 by chris007de