AllYarnsAreBeautiful/ayab-firmware #4: Integrate KH-270 parameters to firmware

chris007de, 2016-11-26:
Do a proper integration of the KH-270 parameters to the firmware, do beautification and code restructuring where necessary.

Sturla22, 2020-06-11:
For future reference: see [branch kh_270](https://github.com/AllYarnsAreBeautiful/ayab-firmware/tree/kh_270)

t0mpr1c3, 2020-07-28:
Is the [current state of the kh_270 branch](https://github.com/AllYarnsAreBeautiful/ayab-firmware/commit/b4b1c62d0f52ae26b5ba5bc8f8e8af26c1612125) a working firmware for the KH-270?

chris007de, 2020-07-28:
> Is the [current state of the kh_270 branch](https://github.com/AllYarnsAreBeautiful/ayab-firmware/commit/b4b1c62d0f52ae26b5ba5bc8f8e8af26c1612125) a working firmware for the KH-270?

As far as we were able to test it back in the days, yes.

t0mpr1c3, 2020-07-28:
> > Is the [current state of the kh_270 branch](https://github.com/AllYarnsAreBeautiful/ayab-firmware/commit/b4b1c62d0f52ae26b5ba5bc8f8e8af26c1612125) a working firmware for the KH-270?
> 
> As far as we were able to test it back in the days, yes.

Great! I have a KH-270 so I can test the changes.

t0mpr1c3, 2020-08-26:
Implemented in https://github.com/AllYarnsAreBeautiful/ayab-firmware/pull/38