AllYarnsAreBeautiful/ayab-firmware #6: Implement CRC8 check for cnfLine messages

chris007de, 2016-11-26:


t0mpr1c3, 2020-04-10:
do we want fast CRC with lookup tables, or is the regular loop method OK?

chris007de, 2020-04-10:
> do we want fast CRC with lookup tables, or is the regular loop method OK?

As this operation is not time critical, we can just use the loop method, I propose.

Sturla22, 2020-06-03:
I have an implementation of the CRC8 here: https://github.com/Sturla22/ayab-firmware/commit/421023d7cbb56d14398398e7dcc50ccdc3ae30eb

Are you aware of any work being planned for this on the AYAB-desktop side?  See https://github.com/AllYarnsAreBeautiful/ayab-desktop/blob/0c298ff83e79d0a94bcc22f05b49452fb67bee8c/src/main/python/ayab/plugins/ayab_plugin/ayab_control.py#L725

Can't actually start checking the CRC before receiving it from desktop, hence the ifdef.

t0mpr1c3, 2023-06-17:
I have removed the preprocessor option AYAB_ENABLE_CRC. I suggest this issue can be closed.

closed 2023-06-17 by dl1com