AllYarnsAreBeautiful/ayab-firmware #7: The last row of a pattern is selected twice.

chris007de, 2017-04-16:
_From @Adrienne200 on December 16, 2016 6:17_

It's too easy to knit the last row of your picture twice. If you follow the instructions precisely and switch to NL(non-pattern) when the double-beep sounds you'll be OK, you won't actually knit that extra row in pattern. But if you miss the double-beep and knit another row in fair-isle it will have added an extra row.
I use the attached pattern when I'm counting rows precisely, 1 black selected in row 1, 2 in row 2 etc. Air-knit with the usual settings, (2 color, single, center, Infinite unchecked.) You knit row 9/select for row 10, and the double-beep happens correctly after you knit row 10. But as you knit row 10, what should it select?
IMO it should not have selected the row 10 pattern again, all needles should now be unselected, at B position. I looking at a set of selected needles and wondering if it finished the pattern properly or not.

[rownumbertest 60x10](https://cloud.githubusercontent.com/assets/24358225/21253365/ec1feb86-c312-11e6-9781-1088bde8f848.png)

_Copied from original issue: AllYarnsAreBeautiful/ayab-desktop#165_

chris007de, 2017-04-16:
This ought to be fixed with  https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/master/knitter.cpp#L266 some time ago...let's look again into this :confused:

Adrienne200, 2017-07-09:
0.9Rc1 fw. The last row is no longer selected twice. 
But I'm not quite ready to close this yet, because it affects the knitting sequence, now you need to knit one more row after all the feedback has stopped, with no sound after you knit the last row. That feels wrong, but maybe it's just different and I just need to get used to it. And document it.

chris007de, 2017-07-18:
@Adrienne200 How do we want to proceed with this issue?

Adrienne200, 2017-12-18:
Closing this, the pattern selection is good; if you keep knitting until no more needles are selected the correct rows have been knitted.

closed 2018-01-11 by chris007de