AllYarnsAreBeautiful/ayab-firmware #8: The double-beep at the end of a pattern is not clearly different from the single beep

chris007de, 2017-04-16:
_From @Adrienne200 on December 16, 2016 5:54_

The double-beep was not different enough from the single on the original AYAB, and is even less distinct on the EMSL board. The single sound is slightly longer than before, and merges with the second beep.

_Copied from original issue: AllYarnsAreBeautiful/ayab-desktop#164_

chris007de, 2017-04-16:
We'll test some other settings for the beeps to achieve a better distinguishment between single and double beep.

Adrienne200, 2017-07-15:
The double beep (or long beep) is much clearer now. But the EMSL boards had had an improved sound, it is now back to a plain buzz.  How would it sound with a longer double-sound but leaving the usual sound as it was?

Adrienne200, 2017-12-18:
This can be closed. The sound was put back to the EMS version for the final release of 0.9 and is more clearly different than it was in 0.8.  It occurs after the last row has been selected but before you knit it, which is OK.

closed 2018-01-11 by chris007de