AllYarnsAreBeautiful/ayab-firmware #15: Clunking sound from the solenoids

chris007de, 2018-10-28:
copied from https://github.com/AllYarnsAreBeautiful/ayab-desktop/issues/256

> When you cancel or finish and restart a pattern, there's often a clunking sound from the solenoids as it resets. The original Brother design never did that. I don't expect that it's doing any harm, but users don't expect it and wonder if there's something wrong.

chris007de, 2018-10-28:
Implement a "cancelKnitting" message to relax the solenoids in case of a cancelled knitting.

t0mpr1c3, 2023-06-17:
I'm not quite clear on what causes the clunk and how to prevent it.

jonathanperret, 2024-09-30:
After some investigation, it appears the clunking sound is caused by all solenoids being turned off at once by AYAB as part of its initialization sequence. This causes each solenoid's armature to jump up as it is pushed by the comb spring underneath.

So, first off this is effectively harmless. Individual solenoids get turned off all the time as part of normal operation, whether using Brother electronics or AYAB. Turning all of them off at the same time shouldn't be more dangerous than turning them off one by one.

Second, this isn't heard with Brother electronics because they default to keeping solenoids **on** when not in active use, i.e. even with no pattern loaded, or when a carriage goes beyond the end of the bed and solenoids don't need to drive any needle. This can be verified by monitoring the voltage level on each solenoid while using Brother electronics.

Note that just turning the solenoids on then off again isn't sufficient to reproduce the clanking sound. The solenoids need to be turned on, then the belt must have moved by at least 16 needles so that all the armatures have been "caught" by the solenoids (they can't pull armatures down, they can only hold them down so the rotary cam's movement is essential to bring the armature in proximity with the elecromagnet so that it can be caught), _then_ when you turn the solenoids off, the armatures will rapidly jump back up causing the clanking sound.