AllYarnsAreBeautiful/ayab-firmware #28: Error Handling and indication via API

Sturla22, 2020-06-03:
It seems that error handling can (and should) be improved. At the moment errors are mostly handled by returning early from void functions.

To handle fatal errors I think it would be best to find a way to indicate an error state to the user. This could be done by adding a byte to the indState message that indicates the current fsm state and then adding a error state. AYAB-desktop should then display a warning to the user on errors.

Non-fatal errors could be logged in a queue and read out by a set of new messages on request.

t0mpr1c3, 2020-07-28:
This can be done without breaking the API because any value other than 1 is effectively already interpreted as an error by the host.

t0mpr1c3, 2020-08-25:
I propose that for fatal errors, the device could enter an error state where the `indState` message, including the error code, is sent repeatedly until the device is reset by some means.

t0mpr1c3, 2020-08-26:
I have coded up an error state in the FSM that flashes the LEDs and send an error message via `indState` twice per second.

What should count as a fatal error? Invalid CRC8 checksum?

How should the error state be cleared? Asking the user to reset the device would not require any further coding, but seems a bit drastic.

chris007de, 2020-09-07:
> I have coded up an error state in the FSM that flashes the LEDs and send an error message via `indState` twice per second.
> 
Why do you send the message twice per second? Is there any reason to do it repeatedly?

> What should count as a fatal error? Invalid CRC8 checksum?
> 
For my understanding, an invalid CRC8 checksum should be handled with retransmission of the packet, not with a fatal error. A fatal error would be something which the Arduino cannot handle/recover without manual intervention by the user. The only thing which comes to my mind in such category would be a unknown combination of hall sensor readings, so the carriage cannot be identified unambiguously.

> How should the error state be cleared? Asking the user to reset the device would not require any further coding, but seems a bit drastic.
For now, manual reset would be a good fallback. But it definitely depends on the error itself. For my example of a non-identifiable carriage, a reset probably wouldn't solve the issue.

t0mpr1c3, 2023-05-15:
https://github.com/AllYarnsAreBeautiful/ayab-manual/pull/13

t0mpr1c3, 2023-06-03:
Closing as the firmware side of this issue is complete. How the host deals with errors can be raised in the desktop repo if necessary.

closed 2023-06-03 by t0mpr1c3