AllYarnsAreBeautiful/ayab-firmware #29: Include parameters for various machine types into single firmware image

chris007de, 2020-06-03:
Currently, we build two different binaries to be compatible with different machine families (KH910 and KH930). These parameters do not have to be chosen at compile time. They rather can be chosen at the start of the knitting process.
Therefore I suppose to create a "monolithic" firmware, which integrates both machine types and provides an additional parameter to choose from them in the reqStart message.

(I started working on this some time ago in 887254c74facf16c9933ff427dcadf08c0f7ae6e)

When this is done, we can also include the parameters for KH270 (#4)

Sturla22, 2020-06-11:
I have a WIP on this here: https://github.com/Sturla22/ayab-firmware/commit/90bf2cd607550af9618845a838f1d733e919d0ee

Need to test on the KH910-type machine I have access to and then it would be good to get someone with a KH930-type machine to test as well. I'll submit a PR once I have tested on HW myself.

t0mpr1c3, 2020-06-11:
^^^ I can run tests on 940 & 270.

t0mpr1c3, 2020-07-22:
The UI side of this is ready to go.

Sturla22, 2020-07-22:
That's great Tom! 
I haven't been able to get my latest changes to work with the KH910 machine yet, not really sure what is wrong.

t0mpr1c3, 2020-08-26:
Implemented in https://github.com/AllYarnsAreBeautiful/ayab-firmware/pull/38

clholgat, 2021-07-01:
I've got some time this weekend and I want to take a look at this.

Was there a plan for selecting which machine to use? From @Sturla22 's WIP PR, it looks like y'all were planning to pass in that parameter from the desktop software.

chris007de, 2021-07-01:
> Was there a plan for selecting which machine to use? From @Sturla22 's WIP PR, it looks like y'all were planning to pass in that parameter from the desktop software.

Correct, starting from 1.0.0, the machine type is chosen in the desktop software. The desktop side has already been implemented by @t0mpr1c3 in https://github.com/AllYarnsAreBeautiful/ayab-desktop/issues/139

t0mpr1c3, 2023-06-17:
This is done now, isn't it?

closed 2023-06-17 by dl1com