AllYarnsAreBeautiful/ayab-firmware #33: Checklist for 1.0.0

t0mpr1c3, 2020-07-28:
What should we prioritize for the next version?

I know people have different things they see as important. For what it is worth, these are my first thoughts on the subject (from https://github.com/AllYarnsAreBeautiful/ayab-desktop/issues/401):

The point of a 1.0.0 release (with semantic versioning) is that we are signalling to users that the API is stable. That is to say, updates to the desktop software are not dependent on updates to the firmware, and updates to the firmware are not dependent on updates to the desktop software. So the most important thing is to get the serial communication protocol right.

Moving to the unitary firmware (AllYarnsAreBeautiful/ayab-firmware#29) requires a change in the serial protocol, so it's a priority to get that sorted out. Integrating support for the KH-270 (AllYarnsAreBeautiful/ayab-firmware#4) will be a big part of that.

Incorporating error messages into the serial protocol (AllYarnsAreBeautiful/ayab-firmware#28) should be possible to do in a way that it is backwards compatible with whatever protocol we adopt. That is to say, the new protocol should be extensible to a reasonably wide range of informative error messages. At the moment that is not the case, but it is an easy fix.

Garter carriage support (AllYarnsAreBeautiful/ayab-firmware#2) is really important but it seems like there are technical challenges that have not been worked out yet. The first question I have is whether supporting the garter carriage requires changes to the serial protocol.

Starting the carriage on the right (AllYarnsAreBeautiful/ayab-firmware#31) seems like low hanging fruit with a big upside because it will enable new knitting modes. But again, the first priority is to figure out whether there are implications for the serial protocol. If not, we can let it slide for the time being.

chris007de, 2020-07-28:
Agreed.
Not to forget https://github.com/AllYarnsAreBeautiful/ayab-desktop/issues/161, which is not API related, but should be investigated nevertheless.

t0mpr1c3, 2020-07-28:
> Agreed.
> Not to forget [AllYarnsAreBeautiful/ayab-desktop#161](https://github.com/AllYarnsAreBeautiful/ayab-desktop/issues/161), which is not API related, but should be investigated nevertheless.

Yes -- to my mind that is a firmware bug, and a significant one, so definitely relevant to this discussion.

t0mpr1c3, 2020-08-24:
One further change to the API that we might consider:

In addition to having the user tell the device what knitting machine is being used, should we allow the user to specify which carriage is in use as well? The knit carriage is the obvious default value. If the device cannot verify the identity of the carriage from the Hall sensor data, it should halt and return an error value.

The changes this would entail for the firmware are:

1. The API messages `reqStart` and `reqTest`, and concomitant functions `startKnitting()` and `startTest()`, need an extra parameter to specify the carriage.
2. Conditions for how the carriage can be verified from Hall sensor data need to be coded and evaluated.
3. The API needs additional message types from device to host to send the error message, and from host to device to acknowledge and/or clear the error state.
4. An error state and associated transitions should be added to the FSM.

3 and 4 are things that we consider adding anyway for other kinds of errors that might arise.

X-sam, 2023-04-08:
Closing this ticket as it's a less-expressive form of the [new release project view](https://github.com/orgs/AllYarnsAreBeautiful/projects/1/views/1). 
@t0mpr1c3 are the notes in [your last comment](https://github.com/AllYarnsAreBeautiful/ayab-firmware/issues/33#issuecomment-679371885) captured in another ticket, or should I copy them to one?

closed 2023-04-08 by X-sam

t0mpr1c3, 2023-04-23:
> Closing this ticket as it's a less-expressive form of the [new release project view](https://github.com/orgs/AllYarnsAreBeautiful/projects/1/views/1).
> @t0mpr1c3 are the notes in [your last [...]

I think it's superceded by recent discussions about carriage detection in the firmware repo.