AllYarnsAreBeautiful/ayab-firmware #48: Garter carriage drift

clholgat, 2022-10-11:
Now that my carriage detection logic made it into the 1.0 release branch, I attempted to go back and resolve some alignment issues only to find further problems.

It looks like there is m_position drift during regular garter carriage usage and I'm trying to understand why and how. 

I added an early out to the carriage selection logic to avoid accidentally resetting the carriage and position with the garter carriage magnet setup. So m_position is only initialized once when the carriage is first detected, after that it's only incremented or decremented. I think that's part of the problem. When using other carriages, you'll often accidentally pass the hall sensors and have m_position reset to something that we know is right.

I added some debugging messages to the knitter, keeping in mind that attempting to log anything slows down operation and can cause its own problems. m_position slowly creeps up when it shouldn't.

I think this is happening during turnaround. I think the logic that reads the pixel buffer and writes it to memory takes a little too long and prevents a few of the encoder's increments/decrements and we wind up a few pixels off. 

I explored tearing during the isrs but all of the important variables are uint8's so that shouldn't happen. I briefly considered memory corruption but I don't see any rogue memory access that looks problematic. 

Anyway, I'm still getting comfortable with the codebase. Here's a draft PR of my debugging code if y'all want to have a look. The diff is not right, I'll fix it later https://github.com/clholgat/ayab-firmware/pull/1

Has anyone seen this before?

clholgat, 2022-10-23:
I'm seeing this in my testing on the 270 as well. I was confused about what's going on because the stitches were mostly correct only to shift over after about 8 rows. It felt error corrected. But after some brainstorming with Adrienne at a meetup, I think I know what's going on.

The solenoids are still being set correctly as long as you're within the right section of them. The firmware thinks your carriage is a few needles off but the solenoids are mostly correct because they're being set a few ahead of the carriage. The few duplicated stitches that I'm seeing are actually bad state being written as if it were part of the next section of solenoids because the firmware thinks the carriage is already over there.

I think this is rarely seen during DBJ and lace carriage work because the carriage crosses the hall sensor on a regular basis and it's position gets reset so it only has the opportunity to get out of sync by a few stitches and the solenoids are still set correctly.

The logic to set the solenoids is inefficient. Right now we're going pixel by pixel, resetting one pixel in the solenoid buffer, and writing the whole buffer to the solenoids every time the carriage moves. We should be able to batch these, write the next 16 pixels to the solenoid buffer and write that to the solenoids only after you cross the threshold to the next set of 16 needles. 

We're also resetting pixels outside of the needle range which might be contributing to the drift during turnaround: https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/v1.0-dev/src/ayab/knitter.cpp#L284

I don't know if that'll speed up the whole process enough that we get less drift but it's worth a try. I'm going to do some prototyping. 

I did some research and it turns out that Arduino will queue interrupts but only one of each type so at most we get one lost carriage motion back. 

This firmware is running up against the limit of what an Arduino is capable of. I'm going to tinker with refactoring for a while but the other option I'm considering are other control boards that are pin compatible with the Arduino so I can get some real multithreading or potentially porting to Raspberry Pi with a breadboard to bridge the gap for now. If that happens, it'll be in a fork.

t0mpr1c3, 2022-10-23:
It's great that you are making progress on this longstanding problem. I hope there is a software solution!

chris007de, 2022-10-24:
> It's great that you are making progress on this longstanding problem. I hope there is a software solution!

Same here, thanks for putting effort in solving this "mystery" :+1: 

Some notes:
- Please keep this in mind while debugging: The parameters for the 270 operation have been found out in a single night of reverse engineering in a garage a few years ago, as we only had this single opportunity to get our hands on a 270. We have no experience about the correctness of these values.

> This firmware is running up against the limit of what an Arduino is capable of.

- I'd assume that the original brains of the machine had about the same or even less performance than an Atmega328, so it _should_ be enough for us as well, but can't tell for sure. Dealing with a multithreading in such a realtime system could probably contain some pitfalls as well. As you describe, there are definitely opportunities to improve the performance of the code (efficiency in solenoid setting, (re)setting unnecessary needles, ...), I think it's definitely worth to improve this as a first step, perhaps this is already the key to avoid race conditions which could lead to undesired behavior. Perhaps there is also some way to intentionally trigger the undesired behavior which would make it easier to observe and debug it?
Anyway, please feel free to refactor anything which might block the road for you :)

Again, thanks a lot! Please let me know if I can support in any way.

clholgat, 2022-10-29:
I've found that the 270 support is mostly correct, it just needed better carriage detection, I'll send out a PR when I'm happy with it.

The original machine couldn't do anything close to what it can with AYAB, hence the increased hardware requirements.

I'm familiar with the pitfalls of multithreading, I've been working as a software engineer for a decade now.

Don't underestimate my desire to make really weird sweaters: https://imgur.com/gallery/WPiQxrg

jpcornil-git, 2023-01-17:
Could you try to add before [L90 of encoder.cpp](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/master/encoders.cpp#L90)/the if statement the line below (same as in Encoders::encA_rising() but with Left/Right swapped)
```python
m_direction = digitalRead(ENC_PIN_B) ? Left : Right
```
The unknown is the behavior of ENC_PIN_A and ENC_PIN_B (V1 & V2 [here](http://hannahnapier.co.uk/wp/wp-content/uploads/2015/02/encoder-signals.svg)) when there is a change of direction. Assuming ENC_PIN_A/V1 behaves always as a clock (top signal below) then ENC_PIN_B/V2 could change before/after the negedge as illustrated below for a right->left transition.

With the current implementation (top diagram) it will only start to count down after the next posedge while with the proposed change it will already start after the next negedge => e.g. extra -1/ no drift 

[image]

AmandineCornil, 2023-01-17:
Just to add info on @jpcornil-git comment; it has been tested successfully with a KG88 on a KH910 machine.

chris007de, 2023-01-19:
@clholgat Could you verify the proposal of @jpcornil-git as well? Thanks for looking into this, very sharp observation of the behavior!

mwhooker, 2023-02-12:
Hi all, I'd like to take a look at this issue, too. I believe I ran into it while making a repeating-pattern test piece, where the patterns got offset.

I am monitoring the rotary encoder position on a second arduino, and am running the code on https://gist.github.com/mwhooker/b6aab659e9ade125c418efd104767eca. I quickly notice that the `Encoders.m_encoderPos` value drifts from the value of my arduino.

However, once I apply @jpcornil-git's patch, the drift disappears.

jpcornil-git, 2023-02-12:
I made another drawing to better illustrate the issue. 

EncA & EncB states are directly/mechanically related to the carriage position, i.e.  change of the carriage position should always align with the same EncA & EncB states for both directions.

In the drawing below and with the current implementation (top), you can see  that for a right->left turnaround:
- (top left) when the change of direction occurs after a EncA negedge everything is fine; negedge becomes a posedge after turnround, direction is therefore updated and next negedge counts down at the right time.
- (top right) when the change of direction occurs after a EncA posedge something wrong happens; posedge becomes a negedge after turnaround, direction is therefore **NOT** updated until the next posedge and position counts down only after the following negedge, i.e. too late.

If the position is also updated upon EncA negedge (bottom drawings), position is correct in both cases.

[image]

For a left->right turnaround, there are no issues as direction is always correct at the first posedge/when position has to be incremented with both current (and patched) implementation, i.e. drift is always to the right.

dl1com, 2023-02-13:
Very comprehensive, thanks for the effort!
@jpcornil-git would you mind to create a pull request for your change?

jpcornil-git, 2023-02-13:
Sure, see PR #50

clholgat, 2023-02-19:
Preliminary tests are looking good over here. I'm going to try a longer and bigger piece tomorrow. I'll let y'all know how it goes!

AmandineCornil, 2023-02-19:
Dear all. I'm curious to know if you succeed to have an auto-detection of your G carriage. While debugging the drift issue with @jpcornil-git, the G carriage was not detected as G but as L or K depending the magnet used for the init (logic I'd we consider the magnets individually). We end up but forcing G in the code but not really convenient.

Hope your long piece (@clholgat) will be OK. I only tested 20 rows. Let is know :-)

clholgat, 2023-02-20:
My big test went well. I'm not seeing any drift after a few hundred rows. I'm comfortable calling this fix good.

dl1com, 2023-02-20:
Merged PR #50 - we'll have to rebase branch v1.0-dev from master so the fix arrives there as well.

Thanks a lot for the investigations @all

closed 2023-02-20 by dl1com

t0mpr1c3, 2023-05-02:
> Don't underestimate my desire to make really weird sweaters: https://imgur.com/gallery/WPiQxrg

Ursula is my favourite character in Little Mermaid. Just saying.