AllYarnsAreBeautiful/ayab-firmware #49: Looking for help installing software onto AYAB / ARDUINO on Mac

robrobss, 2023-01-24:
I have successfully soldered an AYAB shield onto an arduino, but since I have no expertise in coding, etc.,  I am struggeling to get the software onto my arduino. I have a downloaded silabs onto my MAC, but AYAB still will not open. How can I get the software onto the arduino ayab shiled?

festwertspeicher, 2023-02-25:
You can flash the firmware to the arduino using the AYAB software. No need to install anything else.
You can download the latest Version (0.95) [here](https://github.com/AllYarnsAreBeautiful/ayab-desktop/releases/download/v0.95/AYAB-v0.95.dmg)

If you can't open the AYAB Software normally try this:
open your terminal
and go to the destination of ayab (in this example it is in the Applications folder)
`cd /Applications/AYAB.app`
next use this command to open the software
`./Contents/MacOS/AYAB`

After that just plug in your arduino and select 'tools' -> 'flash firmware'.
In the new window select the USB port with your arduino, select your machine (e.g. KH910) and click the button to flash the firmware. you just have to do this step once.

t0mpr1c3, 2023-06-03:
Close as finished.

closed 2023-06-03 by t0mpr1c3