AllYarnsAreBeautiful/ayab-firmware #51: G carriage auto detection

AmandineCornil, 2023-02-19:
Dear all. I'm curious to know if you succeed to have an auto-detection of your G carriage. While debugging the drift issue with @jpcornil-git, the G carriage was not detected as G but as L or K depending the magnet used for the init (logic if we consider the magnets individually). We end up by forcing G in the code but not really convenient.

jpcornil-git, 2023-03-13:
Just ran some tests with L/K & G (KG88 & 93) carriages on a KH910 and logged sensor data.

Signals illustrated herebelow correspond to the following test scenario:
- Crossing the left hall sensor while going left then switch to right
- Crossing the left hall sensor while going right
- Crossing the right hall sensor while going right then switch to left
- Crossing the right hall sensor while going left
- Crossing the left hall sensor while going left then switch to right
- Crossing the left hall sensor while going right

My observations:
### L carriage:
- No detection possible with the right hall sensor (no signal)
- Hall sensor is active for more than one needle/encA cycle. Given that position is held in reset/at 28 during that period, the position on the left side of the hall sensor is shifted by a variable amount compared to what it should be (not dramatic but should be cleaned out). 
- Toggling between 28 (reset value) and 27 when left hall sensor is crossed in the left direction (this is because [encoders.cpp L96](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/master/encoders.cpp#L96) on encA falling edge interfere with [encoders.cpp L84](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/master/encoders.cpp#L84) on encA rising edge. No issues in the other direction because position increment and reset take place on the same rising edge. Not being monotonic may be annoying/unexpected for detection algorithms and it should be cleaned out as well.

[image]

### K carriage:
- Same issue as the one observed for the L carriage and left hall sensor.
- Similar issues with the right hall sensor (offset & toggling) but it now also cause a shift of one needle when carriage is moving left or right. Again not dramatic (what really matters is for coils to be set before used and there seems to be a tolerance of a few needles between effective and expected carriage position wrt that) but it should be fixed.

[image]

### G Carriage (KG88 illustrated here):
- "G" signature only available from the left hall sensor => detection only possible there
- Similar issues with the hall sensors (offset & toggling) as with L & K carriages but because of the "wider signature", offset is larger
- There is a jump of 23 needles when position is reset/hall sensor is crossed, this matches also the measured physical distance between the two magnets, i.e. center is ~11.5 needles away from the center (reference for L& K carriage)

[image]

I will chew a bit on this but I believe based on my limited tests that:
- Except for m_direction, all state changes should be done on rising/falling edges when moving right/left respectively (from encA perspective and for a given physical position a rising edge when moving right is a falling edge when moving left)
- Carriage detection can only be implemented using the left hall sensor
- Position can be forced at both hall sensors but should only be done when moving towards ends and only when threshold is crossed not continuously as of today (and take G offset into account)
- G carriage detection can be implemented when running left by feeding a small FSM with hall detection events (detect two inverse pulse within 3 needles, reset when over/direction change)

You can also browse waveforms by yourself:
- Install [gtkwave](https://gtkwave.sourceforge.net/gtkwave.pdf); on linux, usually a simple
```bash
sudo apt-get install gtkwave
```
- Load one of the trace from [vcd.zip](https://github.com/AllYarnsAreBeautiful/ayab-firmware/files/10955853/vcd.zip) [File -> Open new tab/window]
- Load ayab_waves.sav to add/format waves as above [File -> Read Save File]

Feel free to comment/correct !

In the mean time I will work on a possible implementation (G detection + a few fixes) using the master branch as reference.

Adrienne200, 2023-03-13:
This all looks very promising, thanks. But I believe I have a correction. 
Early in the AYAB project Chris believed the same as you, that L and G carriage detection is only possible on the left. But that wasn't true for the original Brother electronics, all 3 carriages were detected equally well from both ends. So there may be a place in the code where he caused AYAB to not look for carriages arriving from the right, if you can remove that I think you'll find that you can get detection working from both ends.

Buried in some rambling text in #2 I reported these findings: "On both my 930 and 910, I put the original electronics back and tried test 885 from the service manuals. (Similar test in the 930 service manual and "modified" 910 service manual.) Both detected the lace and garter carriages perfectly, on both the right and the left, according to the test display." And it has to work this way, otherwise you couldn't knit reliably all the way to needle Right 100; it needs to detect the correct carriage coming back in to continue knitting.

Though if your 910 is the first generation model, the one that had the two AA batteries, the original electronics didn't support the garter carriage nor the 885 test, so you won't be able to replicate my test. But I found that swapping out just the electronics unit for the 910 second generation, no other changes, caused the garter carriage to work, so AYAB should work too, even on the oldest 910s.

jpcornil-git, 2023-03-13:
I was already puzzled by the different signal shapes out of the left/right sensors seen from firmware but, given @Adrienne200 comment  saying that it does work on legacy machines, I just measured the right sensor output voltage (CH301 herebelow) and it does indeed change as expected for L/K & G carriages ... while (my at least) arduino firmware is not seeing this (but a rather digital-like signal only going down) => I'll investigate this further (both firmware and hardware).

[image]

jpcornil-git, 2023-03-13:
Hardware connectivity:

FW, Right Hall Sensor ---[Arduino, pin A0]---[Ayab shield, JP5 pin 8]---[Flat cable at the top left of the picture above, 3rd wire starting from the left]

That signal is connected to the ouput of a comparator (uPC339, pin 14 of the DIP-14 package top right of the picture) taking the analog value available at the CH301 test point as *input*, i.e. Arduino doesn't see the analog signal (with continuous positive/negative variations as with the left sensor) but only two values; "K polarity detected or none".

This at least explains what I observed.

I see also that another output of the comparator is connected to the 4th wire of the flat cable which is connected/shorted to ground on the ayab shield  JP5 pin 7 (and could be the "L polarity detected or none") ... I'll investigate tomorrow

jpcornil-git, 2023-03-14:
The 4 comparator outputs are connected to pin 7-8-9-10 of JP5 as illustrated below, i.e. 3 of them are effectively short-circuited to ground on the shield (comparator output is open drain -> safe) and only one (green/K polarity) is connected to Arduino (EOL0R, pin A0).

I connected red signal to JP5.8 and it seems to carry the L polarity indication (transitions visible when L (but not K) carriage crosses the turnmark.

[image]

Next HW version of the shield should fix this to support carriage detection on the right (yellow signals/two other comparator outputs should be understood as well ideally)

jpcornil-git, 2023-03-17:
There is also a missing pullup to me (uPC339 comparator is open drain) -> its output is either gnd/0V or highZ/floating (value is a function of surounding circuitry incl. Arduino input impedance), see test below (added a 10K pullup resistance that I removed at the cursor position):

[image]

=> should be addressed by the next HW rev. as well

t0mpr1c3, 2023-03-17:
Yes not a bad idea to explicitly pull it up although Arduino can and presumably does do it in software, although I can't remember the value off the top of my head it is fairly sensible.

jpcornil-git, 2023-03-18:
Yes, this is correct, you can add something like this to the setup code to fix this issue/tested OK
```c
    pinMode(A0, INPUT_PULLUP);
```

Internal pull resistor is 20-50KOhms from [ATmega2560](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-2549-8-bit-AVR-Microcontroller-ATmega640-1280-1281-2560-2561_datasheet.pdf) datasheet, see p355

t0mpr1c3, 2023-03-18:
The lower value of 10K is preferable IMO and we could go lower if switching speed is an issue.

jpcornil-git, 2023-03-19:
If you go fast, e.g. 200 needles in 2s, this is about 10ms/needle while a 10K pullup on a 100pF load gives you a time constant of 1us => all of this is pretty safe to me, lot of margins.

In the mean time, I  implemented a first auto-detection for the LKG carriages; you can have a first look at the code here [hallsensor.zip](https://github.com/AllYarnsAreBeautiful/ayab-firmware/files/11011616/hallsensor.zip) and simplified the main ISR in encoder.cpp (carriage detection code moved to hallsensor class and processing on posedge/negedge when running right/left respectively for the reason explained above).

It should look like this (right hallsensor section commented out as not usable for the time being).

```c
/*
 * Interrupt Service Routines
 */
void isr_encA() {
  static bool _old_encA = false;

  encA = digitalRead(ENC_PIN_A);
  encB = digitalRead(ENC_PIN_B);
  encC = digitalRead(ENC_PIN_C);

  if (!_old_encA && encA) { // Rising edge
    // Update direction
    m_direction = encB ? Right : Left;

    // Update carriage position & type
    if (m_direction == Right) {
      // Moving to the right, check left sensor
      if (leftHallSensor.isActive(m_encoderPos, m_direction)) {
        m_encoderPos = END_LEFT + 28 + (m_encoderPos - leftHallSensor.position);
        m_carriage = leftHallSensor.carriage;
        // Belt phase for left sensor
        switch (m_carriage) {
          case L:
            m_beltShift = encC ? Regular : Shifted;
            break;
          default: // K & G
            m_beltShift = encC ? Shifted : Regular;
        }
      } else {
        if (m_encoderPos < END_RIGHT) {
          m_encoderPos++;
        }
      }
    }
  } else { // Falling edge
    // Update direction
    m_direction = encB ? Left : Right;
    
    // Update carriage position & type
    if (m_direction == Left) {
      // Moving to the left, check right sensor
//      if (rightHallSensor.isActive(hallValueRight, m_encoderPos, m_direction)) {
//        m_encoderPos = END_RIGHT - 28 - (rightHallSensor.position - m_encoderPos);
//        m_carriage = rightHallSensor.carriage;
//        // Belt phase for right sensor
//        m_beltShift = encC ? Regular : Shifted;
//      } else { 
        if (m_encoderPos > END_LEFT) {
          m_encoderPos--;
        }
//      }
    }
  }

  _old_encA = encA;
}
```
[image]

Detection mechanism identify minimum/maximum of the sensor waveform within 3 needles (per manual) and qualify the carriage type base on that (polarity and extremum order) as well as position of the extremum (optimal sensor position). Detection is performed with the left/right sensor when running right/left, i.e. when running toward the center. 

I'll now port it to the master branch (this is running in my own instrumentation code right now) and issue a PR when ready and tested by someone who knows knitting rather than engineering ;-)

jpcornil-git, 2023-03-25:
G carriage detection has been checked-in on top of the dev branch (used by @AmandineCornil) in the following repo https://github.com/AmandineCornil/ayab-firmware.

Detection mechanism is running within a new HallSensor class (see [hallsensor.cpp](https://github.com/AmandineCornil/ayab-firmware/blob/dev/hallsensor.cpp)) by calling its [isDetected](https://github.com/AmandineCornil/ayab-firmware/blob/dev/hallsensor.cpp#L18) method from a refactored/simplified [encA_interrupt](https://github.com/AmandineCornil/ayab-firmware/blob/dev/encoders.cpp#L38) ISR in encoder.cpp.

G carriage is detected when moving to the right and passing the left sensor with the correct waveform (North/South pole levels & order), i.e. only the left magnet trigger a detection. I commented out the right sensor (running when moving left) code because of the KH910 but it should work on e.g. KH930 (once confirmed OK on other machine I can add a compile flag to make it available selectively).

There are also no carriage type "glitches", it jumps directly from NoCarriage to the correct one and you can change carriage "on-the-go", i.e. passing a different carriage over the left magnet change the carriage type dynamically.

clholgat, 2023-04-16:
I'm going to close this ticket. Hardware discussion is going to: https://github.com/AllYarnsAreBeautiful/ayab-hardware/issues/19

And the software discussion can move to the original ticket: https://github.com/AllYarnsAreBeautiful/ayab-firmware/issues/2

closed 2023-04-16 by clholgat