AllYarnsAreBeautiful/ayab-firmware #52: Carriage detection runs before machineType is set

clholgat, 2023-02-20:
I am working on garter carriage detection in the 1.0 branch and I realized that carriage detection runs before machineType is set.

encA* run during the dispatch() loop: https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/v1.0-dev/src/ayab/encoders.cpp#L153

But the machineType isn't set until Knitter::isReady() returns true after the carriage is set https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/v1.0-dev/src/ayab/knitter.cpp#L184

I'm thinking about adding an Nth state to the FSM before s_init and having the desktop software send down machine type as soon as it makes contact. Thoughts?

t0mpr1c3, 2023-02-27:
I agree, this looks like a bug. Carriage detection should wait until after machine type is set. Adding an extra state to the FSM is probably the best way to acheive that.

jpcornil-git, 2023-03-25:
Are detection thresholds really machine dependent ?

I believe that [carriage_hallvalues.md](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/master/carriage_hallvalues.md) should be corrected based on what I described in #51 (hardware for  the right sensor on KH910 is not correct/usable => KH910/Right entries should be ignored).

From the left sensor [waveforms](https://user-images.githubusercontent.com/40644331/226189106-f340f9da-b449-43f2-a016-14377456c847.png) of a KH910 you can also see that "North" and "South" amplitudes are about the same for L/K/G carriages on KH910. I guess that on KH930 they will be very close (values for the left sensor are about the same already), i.e. that 200/600 as MIN/MAX thresholds should be fine for all carriages on both machines.

+ right sensor should be disabled for KH910 with current hardware as I did [here](https://github.com/AllYarnsAreBeautiful/ayab-firmware/issues/51#issuecomment-1483922095) to implement reliable L/K/G detection.

jpcornil-git, 2023-03-27:
Wrt the original question/to address this issue, should the FSM not be something like:
- Knitter ISR disabled at startup (remove [knitter.cpp L63](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/v1.0-dev/src/ayab/knitter.cpp#L63))
   - Note that in today's implementation it is enabled before variables used by the related ISR are initialized. To avoid unexpected behavior it should move at the end of the init() function (or just removed with the suggested approach)
- s_init: Stay there until reqStart (machine type, start/stop needles, ...) is received, then enable knitter ISR and move to s_ready.
- s_ready: Stay there until (m_carriage != NoCarriage) then move to s_knit.

I would also move [L114-118 in knitter.cpp](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/v1.0-dev/src/ayab/knitter.cpp#L114) from ISR to the main code, e.g. to the fsm dispatch() function, to avoid nasty bugs due to an interrupt firing and changing Knitter variable(s) between two lines in the main code. Such handover between ISR/Encoder and main/Knitter context should also be considered as a critical section and protected by disabling/enabling related ISR before/after.

t0mpr1c3, 2023-03-28:
> Are detection thresholds really machine dependent ?

Perhaps Schmitt trigger logic on an interrupt pin could work as well or better than the analog sensor.

jpcornil-git, 2023-03-28:
> > Are detection thresholds really machine dependent ?
> 
> Perhaps Schmitt trigger logic on an interrupt pin could work as well or better than the analog sensor.

You have to encode three states (North/South/None) => to implement that you have to use two external comparators (and two inputs to the MCU) similar to what they did for the right sensor on the KH910 or stick to an analog input with FW implementing the comparaison. Last is most flexible (but may expose machine dependencies to FW) while former is better if already implemented on the original machine (to hide machine details)

For  Ayab, I would minimize external HW/use raw inputs and give FW maximal flexibility/role.

closed 2023-05-18 by clholgat