AllYarnsAreBeautiful/ayab-firmware #54: Unit tests broken

dl1com, 2023-02-26:
see PR #53 and https://github.com/AllYarnsAreBeautiful/ayab-firmware/actions/runs/4270300471/jobs/7442195003

Fix seems to be to update Googletest dependency to > 1.11.0

dl1com, 2023-02-26:
Short test with https://github.com/google/googletest/archive/refs/tags/release-1.11.0.zip shows that this solves the current issue, but yields a new issue within 3rd party code (dangling pointer warning in SoftI2CMater library...).

This has to be fixed first...

```
[ 94%] Building CXX object CMakeFiles/ayab_test_Mega.dir/mocks/fsm_mock.cpp.o
/home/chris/projects/allyarnsarebeautiful/ayab-firmware/libraries/SoftI2CMaster/SoftI2CMaster.cpp: In member function ‘void SoftI2CMaster::setPins(uint8_t, uint8_t, uint8_t)’:
/home/chris/projects/allyarnsarebeautiful/ayab-firmware/libraries/SoftI2CMaster/SoftI2CMaster.cpp:92:18: error: storing the address of local variable ‘port’ in ‘*this.SoftI2CMaster::_sclDirReg’ [-Werror=dangling-pointer=]
   92 |     _sclDirReg   = portModeRegister(port);
/home/chris/projects/allyarnsarebeautiful/ayab-firmware/libraries/SoftI2CMaster/SoftI2CMaster.cpp:76:13: note: ‘port’ declared here
   76 |     uint8_t port;
      |             ^~~~
```

dl1com, 2023-02-26:
@X-sam  thanks a lot for the PR.

As you already mentioned in the PR, tests are able to run again, but many of them fail.
Seems that in most cases it's caused by following or similar errors:

```
/home/runner/work/ayab-firmware/ayab-firmware/test/test_tester.cpp:85: Failure
Actual function call count doesn't match EXPECT_CALL(*serialMock, write(_, _))...
         Expected: to be called once
           Actual: never called - unsatisfied and active
/home/runner/work/ayab-firmware/ayab-firmware/test/test_tester.cpp:86: Failure
Actual function call count doesn't match EXPECT_CALL(*serialMock, write(SLIP::END))...
         Expected: to be called once
           Actual: never called - unsatisfied and active
[  FAILED  ] TesterTest.test_setAllCmd_success (0 ms)
```

jpcornil-git, 2023-02-28:
This is expected given [encoder.cpp L192](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/v1.0-dev/src/ayab/encoders.cpp#L192) addition, i.e. extra read of ENC_PIN_B
=> need to add a line [this]( https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/v1.0-dev/test/test_encoders.cpp#L173) in test_encoders.cpp for all tests creating a falling edge for ENC_PINB and calling encoders->encA_interrupt().

closed 2023-03-13 by dl1com