AllYarnsAreBeautiful/ayab-firmware #62: Tests don't run on Mac?

X-sam, 2023-04-08:
The test suite fails to run on macOS.

t0mpr1c3, 2023-05-02:
Maybe a desktop issue? I think we need a little more triage.

dl1com, 2023-05-18:
What's the status of this issue? From the description, I am not sure what exactly this is about...

t0mpr1c3, 2023-06-03:
Closing as test issues have been raised in the desktop repo.

closed 2023-06-03 by t0mpr1c3

VIPQualityPost, 2023-07-04:
trying to built tests on Mac still fails:
```
fatal error: too many errors emitted, stopping now [-ferror-limit=]
97 warnings and 20 errors generated.
make[5]: *** [CMakeFiles/arduino_mock.dir/src/ArduinoMockAll.cc.o] Error 1
make[4]: *** [CMakeFiles/arduino_mock.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [arduino_mock/arduino_mock/src/arduino_mock-stamp/arduino_mock-build] Error 2
make[1]: *** [arduino_mock/CMakeFiles/arduino_mock.dir/all] Error 2
make: *** [all] Error 2
```
most of the failures are related to `gtest`

VIPQualityPost, 2023-07-05:
OK, it seems the issue is somehow related to picking bad/ old compiler versions. 
I'm not sure how, CMake finds the right compilers: 
```
-- The C compiler identification is AppleClang 14.0.3.14030022
-- The CXX compiler identification is AppleClang 14.0.3.14030022
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE 
```

But when building it gives warnings related to C++11 extensions, and some errors like this:
`error: 'auto' not allowed in function return type` which are features added in C++14.