AllYarnsAreBeautiful/ayab-firmware #86: Knits one row, then dies

t0mpr1c3, 2023-06-16:
Documented in https://github.com/AllYarnsAreBeautiful/ayab-desktop/issues/525

t0mpr1c3, 2023-06-17:
The early termination of the knitting operation occurs because after requesting the first line, the firmware skips to requesting an out of sequence line number. I haven't figured out why this happens yet.

dl1com, 2023-06-19:
With a latest build of Desktop and Firmware (compiled using platformio), communication is reliable and I can't repro the error. Disclaimer: using the DBG_NOMACHINE compile flag, no actual machine attached.

t0mpr1c3, 2023-06-19:
I'm wondering if it's something to do with GARTER_SLOP because @clholgat didn't see it using the GK. Should we be adding GARTER_SLOP for the other carriages?

clholgat, 2023-06-19:
I was able to repro with the K carriage on my 910 with the latest build here: https://github.com/AllYarnsAreBeautiful/ayab-desktop/releases/tag/test230618

But I was not able to repro with a local build of the latest 1.0 branches for desktop and firmware. 

Is it possible that something is wrong with our github builds?

t0mpr1c3, 2023-06-19:
Possibly. How are you making your local builds?

clholgat, 2023-06-19:
I'm running build.sh

t0mpr1c3, 2023-06-20:
> I'm running build.sh

So does the git runner script.

zelatyna, 2023-06-25:
Yesterday tested with [test230618](https://github.com/AllYarnsAreBeautiful/ayab-desktop/releases/tag/test230618) desktop build and still the sam issue - after 1 row of knitting, it dies

Today tested with local firmware build and could not reproduce, tested few times with test pattern 40x40 and it finished knitting fine. I made the local build with platform io.

dl1com, 2023-06-26:
test230626 is the first firmware built by the GitHub actions workflow using PlatformIO.
It's included in the Desktop build with the same tag, let's see how it behaves.

t0mpr1c3, 2023-06-27:
I'll do a test run on 910 later today.

Adrienne200, 2023-07-02:
This issue is no longer present in build test230626's Desktop/firmware. Multiple rows can now be knitted. I'm also closing Desktop issue #525.  Nobody seems to be entirely sure what caused the problem nor what exactly fixed it. individual developer builds were not failing, but the release builds prior to June 26th 2023 were failing. The firmware of test230626 was built with PlatfomIO, maybe that's what fixed it.

closed 2023-07-02 by Adrienne200