AllYarnsAreBeautiful/ayab-firmware #87: cnfInfo needs to report actual firmware version

t0mpr1c3, 2023-06-16:
FW_VERSION_MAJ and FW_VERSION_MIN in com.h need to correspond to the actual firmware version.

Documented in https://github.com/AllYarnsAreBeautiful/ayab-desktop/issues/536

dl1com, 2023-06-16:
I propose something like this script to be called beforehand building the software to use the current git tag as software version automatically

https://github.com/bt-trx/firmware/blob/master/scripts/preBuild.py

t0mpr1c3, 2023-06-17:
Yes we should be able to recover the version number from the git tag.

dl1com, 2023-06-30:
I'd like to add the requirement to mark a firmware build as "dirty", so it is possible to identify non-genuine firmware builds in the logfile.

Assuming that we extract the firmware version from the git tag using `git describe --tags`, we get information if the code has been changed since being tagged. This information should be included into the firmware as well, so we can distinguish a firmware "1.0.0" from a firmware "1.0.0, but I tried something, please beware"

t0mpr1c3, 2023-06-30:
Perhaps the manifest should specify a git checksum for the "clean" version, so that altered builds can be flagged.

Non-numeric tags require patching the fbs version -- which it totally feasible, but a bit of a pain.

dl1com, 2023-07-06:
> I'd like to add the requirement to mark a firmware build as "dirty", so it is possible to identify non-genuine firmware builds in the logfile.
>
> Assuming that we extract the firmware version from the [...]

Currently I am thinking about whether a simple "dirty" flag gives enough information when evaluating a log file.
As we have now the chance to shape APIv6, perhaps it would be useful to enable a field of <to be defined> bytes length, where we can send the suffix of a version number, for example it's a 1.0.0-rc1 firmware. Only having a "dirty" flag hides the information that it was "rc1", it could be virtually any firmware running.

So I am currently thinking of changing the flag into a field of x bytes length. (Perhaps the maximum length of a git describe --tags output suffix, which is `-x-y`, where x is the amount of commits since the tagging and y is the shortened git hash of the current commit)

Any thoughts on this?

t0mpr1c3, 2023-07-06:
Semver 1.0 allows pre-release suffices. Semver 2.0 adds build suffices after that as well. So, are we planning to support semver 1.0 or semver 2.0?

t0mpr1c3, 2023-07-06:
8 bytes for the pre-release suffix and 8 for the build suffix should be enough?

dl1com, 2023-07-24:
Short wrapup:
- With the Pull Requests merged we now have the possibility to have a 16 character suffix (excluding the hyphen between patch level and suffix).
- So example for maximum length of the suffix is: 1.0.0-1234567890123456 (Major, Minor and Patch versions are uint8_t)
- It's automatically baked into the firmware when building running the preBuild.py script.
- The version name is taken from `git tags --describe`.
- In case it's too long, it gets truncated.

closed 2023-07-24 by dl1com