AllYarnsAreBeautiful/ayab-firmware #89: Publish pages

t0mpr1c3, 2023-06-17:
The repo settings need to be changed so that the doxygen makedocs on branch gh-pages get published. Aftter this is done, they should be visible at https://ayab-knitting.com/ayab-firmware/html/index.html

dl1com, 2023-06-18:
Do you know which settings we need to make it work?
Besides that: Shouldn't this step only run i.e. when making a release, not on every PR?

t0mpr1c3, 2023-06-19:
From the repository Settings select Pages, and then under "Build and deployment" select Deploy from Branch >> gh-pages / root

Yes, it doesn't need to run every single PR.

t0mpr1c3, 2023-07-04:
close as completed

t0mpr1c3, 2023-07-04:
Could do with a file http://ayab-knitting.com/ayab-firmware/index.html that redirects to http://ayab-knitting.com/ayab-firmware/html/

dl1com, 2023-07-04:
> Could do with a file http://ayab-knitting.com/ayab-firmware/index.html that redirects to http://ayab-knitting.com/ayab-firmware/html/

In the Github pages settings, the only choices are / or /docs. So it should be sufficient to generate the documentation in / or /docs instead of /html.

t0mpr1c3, 2023-07-05:
Is that actually possible in Doxygen? By default it generates the HTML under /html

closed 2023-07-05 by t0mpr1c3