AllYarnsAreBeautiful/ayab-firmware #108: Some more instruction for building with platformIO?

Tuuleh, 2023-07-03:
I wanted to have a test at #40  - how do I go about doing that?

I've checked out `origin/v1.0-dev` and trying to build there according to the instructions (platformIO on VSCode). Running into trouble with some imports? 

```***************************************************************************
* Looking for Adafruit_MCP23008.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_MCP23008.h"
* Web  > https://registry.platformio.org/search?q=header:Adafruit_MCP23008.h
*
***************************************************************************

 #include <Adafruit_MCP23008.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/uno/src/ayab/main.cpp.o
In file included from src/ayab/tester.h:29:0,
                 from src/ayab/global_tester.cpp:25:
src/ayab/com.h:28:10: fatal error: PacketSerial.h: No such file or directory

**********************************************************************
* Looking for PacketSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:PacketSerial.h"
* Web  > https://registry.platformio.org/search?q=header:PacketSerial.h
*
**********************************************************************

 #include <PacketSerial.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/uno/src/ayab/solenoids.cpp.o
Compiling .pio/build/uno/src/ayab/tester.cpp.o
In file included from src/ayab/knitter.cpp:30:0:
src/ayab/com.h:28:10: fatal error: PacketSerial.h: No such file or directory

**********************************************************************
* Looking for PacketSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:PacketSerial.h"
* Web  > https://registry.platformio.org/search?q=header:PacketSerial.h
*
**********************************************************************

 #include <PacketSerial.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/ayab/main.cpp:28:0:
src/ayab/com.h:28:10: fatal error: PacketSerial.h: No such file or directory

**********************************************************************
* Looking for PacketSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:PacketSerial.h"
* Web  > https://registry.platformio.org/search?q=header:PacketSerial.h
*
**********************************************************************

 #include <PacketSerial.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/ayab/solenoids.cpp:26:0:
src/ayab/solenoids.h:29:10: fatal error: Adafruit_MCP23008.h: No such file or directory

***************************************************************************
* Looking for Adafruit_MCP23008.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_MCP23008.h"
* Web  > https://registry.platformio.org/search?q=header:Adafruit_MCP23008.h
*
***************************************************************************

 #include <Adafruit_MCP23008.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/uno/FrameworkArduino/CDC.cpp.o
In file included from src/ayab/tester.cpp:28:0:
src/ayab/com.h:28:10: fatal error: PacketSerial.h: No such file or directory

**********************************************************************
* Looking for PacketSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:PacketSerial.h"
* Web  > https://registry.platformio.org/search?q=header:PacketSerial.h
*
**********************************************************************

 #include <PacketSerial.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/uno/src/ayab/com.cpp.o] Error 1
Compiling .pio/build/uno/FrameworkArduino/HardwareSerial.cpp.o
*** [.pio/build/uno/src/ayab/fsm.cpp.o] Error 1
*** [.pio/build/uno/src/ayab/global_com.cpp.o] Error 1
*** [.pio/build/uno/src/ayab/global_knitter.cpp.o] Error 1
*** [.pio/build/uno/src/ayab/global_solenoids.cpp.o] Error 1
*** [.pio/build/uno/src/ayab/global_tester.cpp.o] Error 1
*** [.pio/build/uno/src/ayab/main.cpp.o] Error 1
*** [.pio/build/uno/src/ayab/knitter.cpp.o] Error 1
*** [.pio/build/uno/src/ayab/solenoids.cpp.o] Error 1
*** [.pio/build/uno/src/ayab/tester.cpp.o] Error 1
========================================================================= [FAILED] Took 0.49 seconds =========================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Tuuleh, 2023-07-03:
@dl1com that was it, thanks for support :)

closed 2023-07-03 by Tuuleh

Tuuleh, 2023-07-03:
@dl1com actually, running into another issue now, and thought I'd msg you in case you happen to be around instead of opening a new issue. I'll debug this a bit more myself and reach out for more support if needed.

I'm able to build and upload the firmware without any issues. I'm testing together with ayab desktop on the `1.0.0-dev` branch, and somehow the connection doesn't work between the arduino and the desktop software anymore. When I click "start knitting", the row counter in the desktop software doesn't advance, all needles other than edge needles in B position get pushed to D position, and I don't hear the usual "click" sound it makes when it engages with the machine. The LED light on the ayab shield is on. Could you tell me if there are some steps I've forgotten about for testing? I've got power turned on to the machine and the machine is turned on.

I haven't used AYAB in months and didn't try the connection before I flashed the new version of the firmware, so now I can't be sure if it's because of the firmware change, or if it's because of e.g. some hardware issue

zelatyna, 2023-07-03:
@dl1com I made a small PR to add this to README as I also came across this issue and I think does not harm to have more detailed instructions