AllYarnsAreBeautiful/ayab-firmware #110: Lace mispatterning on v1.0-dev with KH-940

Tuuleh, 2023-07-04:
It seems that the duplicate needle selection described in #40 is no longer an issue, but there's now another problem where needle selection does not work with the lace carriage. I've had two instances where one needle has not been selected when pushing the carriage to the right (which may have been a fluke), but consistently run into an issue with needles not being selected when pushing the carriage to the left. 

In case of one test pattern, no needles were selected when pushing to the left. In case of using the test pattern from #40, no needles left from the middle of the pattern are selected when pushing the carriage to the left. I've tested with two different lace carriages with identical results (it's very consistent in what doesn't get selected with more or less 0 variance), and I've also tested by offsetting the lace pattern from the middle, in which case the behavior is the same (needles from the left half of the pattern are not selected when pushing to the left). 

Pushing the carriage to the right seems to lead to correct needle selection. 

Here's a video of the issue: https://photos.app.goo.gl/FJKa48fDyYJTwiCw6

dl1com, 2023-07-14:
Could you please verify that this is/is not the same behavior as described here: https://github.com/AllYarnsAreBeautiful/ayab-firmware/issues/40#issuecomment-1626776821

Tuuleh, 2023-07-15:
@dl1com I opened a new issue because this behavior when testing with the same lace pattern is different for me than what was happening before the v1.0 release. When knitting with 0.95, the mispatterning would happen after knitting more rows. It'd always be the same adjacent extra needles selected. When knitting with 1.0, the mispatterning would begin very quickly, and needle selection when pushing to the left would almost never happen at all. Then at some point, there would be the mirroring behavior described in the updated steps to reproduce under #40 - I don't remember if this was present in 0.95.

I think it might be best to close this one and track under #40. It's likely a similar root cause in the mispatterning between 0.95 and 1.0, just changed by some code that slipped in in between (e.g. the belt shift fixes). 

I'll leave it up to you if you'd like to close this one.

X-sam, 2024-03-30:
let's consolidate on #40

closed 2024-03-30 by X-sam