AllYarnsAreBeautiful/ayab-firmware #156: Need a 'Cancel' command in the API

t0mpr1c3, 2023-09-27:
I'm also concerned that a bug happens when you press `Cancel` in the desktop. The firmware stays in the `Knit` state waiting for the last line of the pattern, and never receives it.

There probably needs to be a `Cancel` command in the API that resets the state to `Init`.

t0mpr1c3, 2024-03-13:
In relation to this:

When knitting/testing is initiated the host sends a request and the device confirms the request. But when testing is ended, the host sends a quit command with no acknowledgment. 

I think the device should acknowledge the quit command, and also acknowledge cancelation of the knitting operation.

jonathanperret, 2024-09-26:
I can confirm this behavior in the current 1.0 beta: after canceling a knitting operation, the last line sent to the firmware is repeatedly selected if you keep doing rows with the carriage in KC position.
Not sure this is critical for 1.0 though.