AllYarnsAreBeautiful/ayab-firmware #176: KH930: Lace carriage not recognized from the right

Adrienne200, 2023-10-31:
The 930 and later machines are able to send a North or South signal from both left and right sensors. The firmware needs to recognize South as a lace carriage

AYAB software version: 0.99 rc1 Aug 6 2023
Computer/OS: Windows 7
Knitting machine: KH930
AYAB hardware: Interface

On a KH930, open any pattern.
Set AYAB to all defaults.
Start the lace carriage from the right.
Expected:
 It should recognize that a carriage is there, and that it is the Lace carriage, and continue accordingly to select row 1.
Actual:
  Nothing happens. 
  Take the carriage all the way across and come in from the left, now it thinks it's on row 1.

(This may never work with the current hardware on a 910, see the diagnosis in fw #51 , but it should be possible on the other machines.)

Adrienne200, 2024-02-20:
This one may be addressed by PR #184

jonathanperret, 2024-10-31:
It was not fixed by #184 . I'm working on this with all the other "starting from the right" issues.

PR #205 (2025-02-08): Fix carriage detection on the right
## Problem

Fixes carriages not being detected on the right on KH930-like machines and KH-270 (fixes #175, fixes #176, fixes #215, fixes #216).

## Proposed solution

On the KH910/KH950, we can't fix detecting Lace or Garter carriages on the right without an AYAB hardware change. But on other machines, the right sensor is just as functional as the left one.

So in this PR, we mostly copy the logic from `encA_rising` into `encA_falling`, and add a special case for the KH910 with its inverted K-only reading.

We also have to change the initialization logic (in [... cut]
closed 2025-02-08 by dl1com