AllYarnsAreBeautiful/ayab-firmware #177: KH930: pattern with 200 stitches, needle L100 is always selected.

Adrienne200, 2023-10-31:
AYAB software version: 0.99 rc1 Aug 6 2023
Computer/OS: Windows 7
Knitting machine: KH930
AYAB hardware: Interface

KH930, K-carriage from the left is almost working perfectly. But with a pattern that's the full 200 stitches wide, it always selects needle L100. This does not happen on a KH910.

Steps:
On a 930.
Open the attached image
Set all UI settings to defaults
Set the K- carriage to KC2 and start from the left.
Set all needles in work.
Air-knit as usual. (As you're using all 200 needles, be extra careful to stop as soon past the sensor as possible and wait for the multiple-beep)
Expected: It should follow the pattern exactly. The end squares are white, so when using KC2 they should be at B.
Actual: Needle Left 100 is selected to B.
[image]

clholgat, 2024-02-11:
I can repro with your 930. Looks like it's only happening when the carriage is moving pretty fast and happens more frequently left to right than right to left. 

I have a few theories. There could be garbage left in the solenoid buffer from the previous row. This could be an issue with the turnaround because the carriage disengages with the timing belt on the edge of the bed and we may not read enough extra needles to indicate the end of a row. So it's playing catchup when it gets to the next row and hasn't already fetched the next row of pixels. 

Time to do some debugging.

clholgat, 2024-02-11:
Hmm nevermind, I was testing with KC1 instead of KC2. I can't repro with KC2 on the latest software. 

Though it is odd that end needle selection isn't working consistently with KC1, though that is probably not an AYAB issue.

X-sam, 2024-03-30:
can't repro so closing for now

closed 2024-03-30 by X-sam