AllYarnsAreBeautiful/ayab-firmware #179: KH910 garter carriage mispatterns on a 190-stitch wide pattern.

Adrienne200, 2023-11-01:
Narrow patterns in the middle of the bed are patterning correctly with the garter carriage on a KH910 starting from the left. But if the pattern goes all the way to the ends, the right-left row adds in lots of extra purls.

AYAB software version: 0.99 rc1 Aug 6 2023
Computer/OS: Windows 7
Knitting machine: KH910
AYAB hardware: Shield

KH910 KG88 garter carriage, starting from the left.
Open either of the attached images
Set all UI settings to defaults, except set L95 and R95 as the start and end needles
Set needles in work from L95 - R95. (The KG88 manual says that it doesn't work on all 200 needles, only out to 95.)
Cast on and knit some stockinette rows with the K carriage to get started. 
Thread up the G carriage and start from the left.
Let it knit at least 6 rows.

Expected: it should be following the pattern exactly.
Actual: The left and center motifs are correct. But over on the right, from about R45 onwards, the right-left rows are not picking up the pattern, instead they are doing roughly alternating bytes of knit and purl.

[image]
[image]
Reading each photo up from the bottom, with right and left as you are looking at the purl side on the machine.
First I did the pattern with the vertical lines. The two bands of garter stitch near the right end should not be there, that area should be all purl.
Then the fish and helicopters, with settings at default (so Knit Side Image is unchecked). The two helicopters are correct, over on the right you can just make out that the left-right rows were trying to make the fish image correctly, but the right-left rows had lost the pattern completely. Then somewhere about R45 the pattern becomes correct again as it continues towards the left.
Then I restarted the same pattern with Knit Side Image checked. Now the fish are correctly over on the left of the bed, and swimming to the left. Same with the center helicopter, facing to the left. Over on the right, again you can maybe see the helicopter facing to the left, with the same errors on the Right-Left rows.
(There's also a whole row of knit stitches where I restarted with Knit Side Image checked. Not sure yet if that is a bug or operator error.)

[image]
[image]

[image]
[image]

Adrienne200, 2023-11-02:
No need to knit all 190 needles to see this. I tried a 60stitch pattern that worked perfectly with the garter carriage in the middle of the machine, but when it tries to do the same pattern on R35 - R95, same errors. The left-right rows are making the pattern correctly, right-left give the same mispatterning. I think it's because the carriage has passed the right hall sensor, and when returning from outside to inside just the North magnet is detected, confusing it. Even though the UI still knows that it's a G carriage. Surprise is that it does somehow recover and do most of the bed correctly.

doo357, 2024-07-25:
I consider this error to be one of the few that makes it fundamentally impossible to use some of the equipment’s functions. The topics recently discussed in the community relate to improving the mechanisms that now also work, albeit not in the best way, but they work. It is now impossible in principle to knit by robot  when rapport width is >160 - neither manually nor with inconvenience. The error is consistently reproduced on 2 types of machines - 910, 930.

clholgat, 2024-07-26:
I'm looking at this but waiting on a new fuse for my GC

doo357, 2024-07-26:
When I knit 200 needles by KG, I remove right magnet in the middle of 1 row. Rapport <160, Release 0.95 Win. I didn't found another way. :(( On 0.99 it not work.

clholgat, 2024-07-28:
Here's where I am in my debugging:

- It's not a carriage detection issue
- It's not an alignment issue
- It only happens when both magnets have crossed the right turn mark

Checking to see what else could be causing this.

clholgat, 2024-07-28:
OMG it's a combination of type coercion and uint8_t overflow.

This would probably happen if you take the main carriage and move it all the way off the bed in the middle of knitting.

Working on a fix.

X-sam, 2024-07-28:
Oh God. Nice catch!

X-sam, 2024-07-29:
Is it in [Knitter::calculatePixelAndSolenoid()](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/v1.0-dev/src/ayab/knitter.cpp#L389)?

clholgat, 2024-07-30:
It is not, it is here: https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/v1.0-dev/src/ayab/encoders.cpp#L137

The garter carriage can advance the belt past position 255 and this +/- 1 coerces the uint8_t to an int so it's not even overflowing correctly.

doo357, 2024-08-01:
Hi! Why for 930 Lmin/Lmax is 200/600, bur Rmin/Rmax is 0/600, sensors is similar!
//                                                                                            KH910 KH930 KH270
constexpr uint16_t FILTER_L_MIN[NUM_MACHINES] = { 200U, 200U, 200U};
constexpr uint16_t FILTER_L_MAX[NUM_MACHINES] = { 600U, 600U, 600U};
constexpr uint16_t FILTER_R_MIN[NUM_MACHINES] = { 200U,   0U,   0U};
constexpr uint16_t FILTER_R_MAX[NUM_MACHINES] = {1023U, 600U, 600U};
May be it is a reason, why right sensor not detected correctly S/N magnet for G carriage?

jonathanperret, 2024-09-26:
Fixed by #196.

closed 2024-09-26 by jonathanperret