AllYarnsAreBeautiful/ayab-firmware #185: Is the long delay at the beginning of the first row still needed?

Adrienne200, 2024-02-20:
AYAB has always needed a delay of several seconds after you cross the left turn mark for the first time, while it waits to be sure that the first row is fully available. We had a brief discussion about it in one of the dev meetings in 2023, but I'm not sure if anyone investigated.
Native Brother mode doesn't have any delay, you can just cross the turn mark and start knitting. So new users often miss that they need to wait, and even experienced AYAB users will sometimes forget.

t0mpr1c3, 2024-02-23:
As of now it is a fixed 2 second delay. I don't think it accomplishes anything, I suspect it is a holdover that we never got rid of.

X-sam, 2024-03-30:
@X-sam to make a test firmware for this and @Adrienne200 to confirm

Adrienne200, 2024-04-21:
The delay is maybe a little shorter now, but it is absolutely still needed. So unless it can be eliminated completely, we still need to document the requirement to hesitate for 2 seconds after crossing the turn mark on the first row. And the reminder sound is less clear now than in 0.95, I'll be documenting that separately. 
It is still possible to get it to completely miss the first row and send the row 1 pattern data in row 2. Just zoom out from the left and shoot across, (as Brother knitters are accustomed to doing) it's possible to get well past the center before AYAB has caught up, so that it sends row 1's data as if it's row 2.

closed 2024-07-30 by dl1com