AllYarnsAreBeautiful/ayab-firmware #204: Needle selection is wrong on KH270, Beta builds.

Adrienne200, 2024-10-09:
AYAB software version: 1.0.0-beta2
Computer/OS: Mac. Intel, Big Sur 11.7.10
Knitting machine: KH270
AYAB hardware: Tried both a shield and an interface

Steps:
1. Make sure you have selected KH270 in prefs, and reflash the firmware.
2. Open the Triangles 60x10 image.
3. Leave all the settings at default.
4. Air-knit, or knit in Fair Isle

Expected: it should select needles looking exactly like the pattern on the purl side, reversed on the knit side.

Actual: The whole pattern is displaced 2 needles to the right, and alternate rows are displaced by 6 needles. It's not entirely consistent, sometimes a sequence of a few rows will be correct (aside from being displaced) but usually it's wrong on alternate rows.

I tried with and without crossing the turn marks every row. No difference.

It was mostly working previous to the Beta builds, I went back to a build from Aug 16th (including flashing the firmware) which I think was one of Jonathan's test builds and it was still good there.

jpcornil-git, 2024-10-11:
6 seems to point to HALF_SOLENOIDS_NUM 

The only place where it plays a role is here 
```c
  m_pixelToSet = m_position - startOffset;

  if (!beltShift) {
    m_solenoidToSet = m_pixelToSet % SOLENOIDS_NUM[static_cast<uint8_t>(m_machineType)];
  } else {
    m_solenoidToSet = (m_pixelToSet + HALF_SOLENOIDS_NUM[static_cast<uint8_t>(m_machineType)]) % SOLENOIDS_NUM[static_cast<uint8_t>(m_machineType)];
  }
```
https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/main/src/ayab/knitter.cpp#L419

Direction doesn't play a role here a priori; starfOffset is 28/16 for left/right respectively but after modulo 12 operation, end results will be the same, i.e. they could be set to any multiple of 12.

... but beltshift is supposed to be "not connected" (even a floating value should be independent of the direction)

I can't find a reason to explain why the pattern is displaced by 2 needles to the right ... the solenoid to set is always the pixel to set modulo 12 -> either at the correct place or shifted by a multiple of 12.

Could you check/confirm that the solenoid connector is well in place, not shifted ?

Adrienne200, 2024-10-13:
My 270 patterns correctly using its native electronics - I checked that patterns are positioned where they should be, relative to the center of the machine. And yes, my number strip is positioned correctly.
I've used two different AYAB hardwares, a shield and an interface. The connectors only go in one position.

jpcornil-git, 2024-10-15:
Could you check in your tests if:
- none of the two magnets crosses the right sensor.
- the first magnet crossing the left sensor is the one at the right of the carriage.

When I do that with my simulator, I see shifted patterns or marginal needle selection (solenoid is set when the needle selection takes place).

I see in the current code that the carriage position is reset everytime a magnet crosses the right sensor in whatever direction to 135 ([END_RIGHT_MINUS_OFFSET](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/main/src/ayab/encoders.h#L75)) at [encoders.cpp#L285](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/main/src/ayab/encoders.cpp#L285)

On the left side, position is set to 5 + 12 = 17 the first time a magnet seen ([END_LEFT_PLUS_OFFSET](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/main/src/ayab/encoders.h#L74) + [MAGNET_DISTANCE_270 ](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/main/src/ayab/encoders.h#L116)) at [encoders.cpp#L199](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/main/src/ayab/encoders.cpp#L199) (detected_carriage is always "Knit" for KH270 and therefore "if" statement is always taken).

135 -17 = 118 = 112 + 6 => shifted by 6 when right magnet of the carriage is crossing the right sensor.

17 seems correct when carriage is engaged from the left and the first magnet crossing the left sensor is the one at the right of the carriage. If the other magnets crosses first (carriage coming from the center before init, not enough to the left at start, ...) , then position is also set to 17 but the carriage is effecti 6 needles further => shifted by 6 as well.

clholgat, 2024-10-16:
I'm going to test this on my 270

clholgat, 2024-10-16:
Yup, I can repro, probably an issue with the offsets, working on a fix.

jpcornil-git, 2024-10-18:
Simulation results;
- ayab-desktop with continuous reporting, i.e. what the FW indicates
- simavr, i.e. what the machine does  (solenoid register bank, needle "painted" ('x'), carriage position ("^"), needle bed, ...) 

Released version (NOK)
- ayab-desktop [ayab_monolothic_uno_desktop.log](https://github.com/user-attachments/files/17431099/ayab_monolothic_uno_desktop.log)
- simavr [ayab_monolithic_uno.log](https://github.com/user-attachments/files/17431155/ayab_monolithic_uno.log)

Proposed/this fix (OK)
- ayab-desktop [ayab_desktop.log](https://github.com/user-attachments/files/17431098/ayab_desktop.log)
- simavr [ayab.log](https://github.com/user-attachments/files/17431154/ayab.log)

dl1com, 2024-10-27:
Solved by #207

closed 2024-10-27 by dl1com