AllYarnsAreBeautiful/ayab-firmware #211: Add proper support for PCF8574 port expander

dl1com, 2024-11-06:
As discussion on Discord revealed, we are using to different types of I2C port expanders on shield and interface.
Currently, we are only supporting the MCP23008 type properly.
They are somewhat compatible in terms of their protocol, so this is not a breaking issue and may have not been an issue in the field so far, but should be fixed to avoid sporadical issues/unwanted behavior.

Support for the PCF part had been added in the past, but was removed accidentially (research done by @jonathanperret):

> Ah, I just did a bit of Git archaeology on that MCP23008/PCF8574 issue and the results are… interesting.
The issue with driving a PCF8574 with code intended for the MCP23008 was very clearly identified and solved (with a patch to the MCP23008 library) by Windell in https://github.com/AllYarnsAreBeautiful/ayab-firmware/pull/11 !
>
 >   "With this new version, the I/O signals on PCF8574-based boards are vastly improved (clean steps, no longer a forest of spikes), as seen on the logic analyzer. This may help to cut down on occasional glitches on PCF8574 based boards."
>
>This is the code that shipped in the 0.95 firmware.
Unfortunately, this customization of the MCP23008 library was overlooked as the code for the libraries was removed from the repository, in https://github.com/AllYarnsAreBeautiful/ayab-firmware/pull/27 .

jpcornil-git, 2024-11-07:
Detection between both devices can be implemented using bit 0 of IOCON register; write 1 -> read 1 on PCF8574 but 0 on MCP23008. Based on that, the device configuration and writes to the address register can be skipped.

jonathanperret, 2024-11-07:
> Detection between both devices can be implemented using bit 0 of IOCON register; write 1 -> read 1 on PCF8574 but 0 on MCP23008. Based on that, the device configuration and writes to the address register can be skipped.

Yes, and the code in https://github.com/AllYarnsAreBeautiful/ayab-firmware/pull/11 can be used as a reference for this approach.

dl1com, 2025-04-01:
Is this already fixed in ayabasync? Not sure if I should flag it as v1.0.0 issue

jpcornil-git, 2025-04-01:
No, but probably easier to implement in ayabAsync as it doesn't depend on Adafruit_MCP23008 library.
Whenever required, I can create another device for pcf8574 here https://github.com/AllYarnsAreBeautiful/ayab-firmware/tree/ayabAsync/lib/devices with the same interface as mcp23008 and instantiate one or the other after detection in the `Solenoids` [constructor](https://github.com/AllYarnsAreBeautiful/ayab-firmware/blob/d9ad53f844c1f5f554c28274945eafbae47ac7ee/lib/components/solenoids/solenoids.cpp#L3C12-L3C21).

VIPQualityPost, 2025-08-24:
Did we end up rolling this into ayabAsync?
When we were at Open Sauce last month, we visited Sequioa Fabrica (a space that focuses on textiles) and they had ordered quite a large amount of PCF8574 boards for converting all of their machines, but ran into this issue.

jpcornil-git, 2025-08-25:
No but it is straightforward to implement whenever required (someone will have to test the fix as I don't have an interface myself)

VIPQualityPost, 2025-08-26:
@jpcornil-git should it target [ayabAsync branch](https://github.com/AllYarnsAreBeautiful/ayab-firmware/tree/ayabAsync) of ayab-firmware or repo on[ your account](https://github.com/jpcornil-git/ayabAsync-firmware)?

jpcornil-git, 2025-08-26:
Agree.

@VIPQualityPost submitted already a PR on ayabAsync-firmware, see https://github.com/jpcornil-git/ayabAsync-firmware/pull/2, and I just created a pcf8574 branch with an implementation proposal to test, see https://github.com/jpcornil-git/ayabAsync-firmware/compare/v1.0...pcf8574

=> we will submit a PR into the official AYAB firmware/ayabAsync branch once the proposal tested/validated

VIPQualityPost, 2025-09-22:
Sorry - I am out of the country for a work trip, I can test this in a week or two when I've returned. I think it is only held back by testing now?