AllYarnsAreBeautiful/ayab-firmware #220: Switching to AyabAsync

dl1com, 2025-03-19:
This issue shall act as an epic to keep track of the necessary tasks in order to switch from the current AYAB firmware ("AyabSync") to @jpcornil-git's implementation AyabAsync.

- [ ] Test mode: "I haven't implemented the current test mode because it uses a terminal-like interface that requires strings to be stored in the firmware (which has limited resources) and parsed by the desktop software, and which also limits the testing capabilities to what has been predefined => It would be better to extend the current API with low-level primitives for manipulating the hardware and implement the test code in the desktop software rather than in the firmware... or perhaps drop it if continuous reporting becomes the norm (you get all the inputs and outputs/solenoid tests can use test patterns)." (reported by @jpcornil-git in Discord)

- [ ] CI: "I haven't ported the current test code because I'm not sure what's next; shouldn't unit tests be limited to certain functions and global tests based on the ayab/simavr simulator (allowing to cover all machines/carts/… much more efficiently than today) that I developped and @jonathanperret (KH910) started to adapt for CI (remove UI and replace by a test ifce)?" (reported by @jpcornil-git in Discord)

jonathanperret, 2025-03-19:
Here’s a link to the project in question: https://github.com/jpcornil-git/ayabAsync-firmware

jpcornil-git, 2025-03-19:
I suppose that the idea would be to create a new branch in ayab-firmware (from root probably given that there is no relation/history between these two codes), i.e. I would park my current repo and work on this one instead ?